rust-embedded/cortex-m-quickstart #15: examples/device.rs doesn't make a whole lot of sense

therealprof, 2017-07-09:
As I've just discovered, SysTick does not generate exceptions by default and requires configuration to do so. Unfortunately the configuration can not be done without direct register manipulation since the SVDs (even the current official ones) lack the description of the required STK_ addresses.

I my opinion examples should always be a good starting point for further tinkering but in the case of SYS_TICK it's really a dead end...

japaric, 2017-07-11:
Have you checked the [API documentation](https://docs.rs/stm32f103xx/0.7.1/stm32f103xx/) of the device crate? [cortex_m::peripheral::SYST](https://docs.rs/cortex-m/0.3.0/cortex_m/peripheral/struct.SYST.html) is re-exported there. That struct not only gives you access to the SysTick registers, but also a high level API to generate periodic interrupts.

I can expand the example.

therealprof, 2017-07-11:
I have, though it was not trivial to find it there. The whole setup is still a bit confusing with some registers being generated on the fly from the SVD files and others not and many of them having all-so-slightly different names than the original ones.

I'm still fighting with setting up the SysTick timer, though. I cannot get it to fire using Rust code while other (C-based) APIs work just fine. The exception vector seems to be set-up alright and the SysTick initialisation is fine as well; most likely I'm missing some magic NVIC initialisation here or there's some conflict in the interrupt handling since all other sources do not use a critical section inside their SysTick handler...

japaric, 2017-07-12:
@therealprof See #16

therealprof, 2017-07-12:
That's where I started from. I have the exact same SysTick and exception setup but the exception is never raised. I'm on a Cortex-M0 BTW, maybe that makes a difference wrt to the interrupt/exception default setup. Working code is fiddling with the Exception priority in the SCB which is what I'm checking out right now.

japaric, 2017-07-12:
I can think of two possibilities: (a) your device has a vendor specific SysTick timer -- SVD files have [a field](https://www.keil.com/pack/doc/CMSIS/SVD/html/elem_cpu.html), vendorSystickConfig, to indicate this scenario; or (b) your device has no SysTick peripheral; this is something I have not heard about before. Maybe check if the writes to the registers are being performed by reading back the registers

therealprof, 2017-07-12:
Hm, I got it working, but not reliably. Seems to be some random crap in the register somewhere.

Some nits about the SYST implementation: fn clear () does not make a whole lot of sense as clearing actually disables the SysTick timer because the exception is triggered when the count changes from 1 -> 0 which is also what causes the reload. Similarly the manuals suggest you should load the counter with the value 1 (for which there's currently no safe API) to ensure that the timer will not start with some offset in the first turn and also the correct value for the reload is actually <desired count> - 1.

japaric, 2017-07-15:
Closing since #16 landed.

@therealprof I have forwarded your feedback on the SYST API in japaric/cortex-m#59

closed 2017-07-15 by japaric