rust-embedded/cortex-m-quickstart #17: Examples don't build

matthewmummert5, 2017-10-01:
```
[matthew@ArchSystem cortex-m-quickstart]$ xargo -V
xargo 0.3.8
cargo 0.23.0-nightly (e447ac7e9 2017-09-27)
[matthew@ArchSystem cortex-m-quickstart]$ rustc -V
rustc 1.22.0-nightly (6f87d20a7 2017-09-29)
```

Following the steps at https://docs.rs/cortex-m-quickstart/0.2.1/cortex_m_quickstart/ results in examples that don't build.

I'm trying to build the example `hello.rs`:

```
//! Prints "Hello, world!" on the OpenOCD console using semihosting
//!
//! ---

#![feature(used)]
#![no_std]

extern crate cortex_m;
extern crate cortex_m_rt;
extern crate cortex_m_semihosting;

use core::fmt::Write;

use cortex_m::asm;
use cortex_m_semihosting::hio;

fn main() {
    let mut stdout = hio::hstdout().unwrap();
    writeln!(stdout, "Hello, world!").unwrap();
}

// As we are not using interrupts, we just register a dummy catch all handler
#[link_section = ".vector_table.interrupts"]
#[used]
static INTERRUPTS: [extern "C" fn(); 240] = [default_handler; 240];

extern "C" fn default_handler() {
    asm::bkpt();
}

```

And I get this error:

```
[matthew@ArchSystem cortex-m-quickstart]$ xargo build --target thumbv7em-none-eabihf
   Compiling vcell v0.1.0
   Compiling cortex-m-quickstart v0.2.1 (file:///home/matthew/Rust/cortex-m-quickstart)
   Compiling bare-metal v0.1.1
   Compiling r0 v0.2.2
   Compiling cortex-m v0.3.1
   Compiling cortex-m-rt v0.3.5
   Compiling aligned v0.1.1
   Compiling cortex-m-semihosting v0.2.0
   Compiling volatile-register v0.2.0
   Compiling stm32f103xx v0.7.5
error: linking with `arm-none-eabi-ld` failed: exit code: 1
  |
  = note: "arm-none-eabi-ld" "-L" "/home/matthew/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.0.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.1.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.10.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.11.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.12.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.13.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.14.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.15.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.16.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.17.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.18.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.19.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.2.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.20.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.21.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.22.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.23.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.24.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.25.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.26.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.27.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.28.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.29.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.3.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.30.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.31.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.4.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.5.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.6.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.7.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.8.o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe.9.o" "-o" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/cortex_m_quickstart-d4d727dd7bcc42fe" "--gc-sections" "-L" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps" "-L" "/home/matthew/Rust/cortex-m-quickstart/target/debug/deps" "-L" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/build/cortex-m-quickstart-f3f9817348225d8c/out" "-L" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-90b1719ebf8fd2a6/out" "-L" "/home/matthew/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-084609a7eabef0b0.rlib" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/libr0-139662ce822171f5.rlib" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_semihosting-d01acf028121f941.rlib" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-b6aa1a4b30b94c14.rlib" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-6d3be0f20e7650c5.rlib" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/libaligned-4ec481963f639147.rlib" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-e7853a3b6bd73682.rlib" "/home/matthew/Rust/cortex-m-quickstart/target/thumbv7em-none-eabihf/debug/deps/libvcell-87f07f6d6fae9582.rlib" "/home/matthew/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-e51a562f0b13f7ca.rlib" "/home/matthew/.xargo/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-d512a12552ca0e4c.rlib" "-Tlink.x" "-Bdynamic"
  = note: arm-none-eabi-ld: 
          The exception handlers are missing. This is likely a cortex-m-rt bug.
          Please file a bug report at:
          https://github.com/japaric/cortex-m-rt/issues
          arm-none-eabi-ld: 
          Invalid '.vector_table.exceptions' section. This is likely a
          cortex-m-rt bug. Please file a bug report at:
          https://github.com/japaric/cortex-m-rt/issues
          

error: aborting due to previous error

error: Could not compile `cortex-m-quickstart`.

To learn more, run the command again with --verbose.

```

Am I doing something wrong? Or is this a bug? I've only been using rust for a few months, so this could be my mistake.

japaric, 2017-10-02:
I can repro this with `rustc 1.22.0-nightly (6f87d20a7 2017-09-29)`, but linking works fine with `1.22.0-nightly (325ba23d5 2017-09-19)`. So this may be a rustc regression. I can't bisect the toolchain to nail down the problem right now so I'll advise you to rollback your toolchain version for now.

pftbest, 2017-10-03:
@japaric msp430-quickstart also doesn't work atm, but it fails with a different error message.

attdona, 2017-10-06:
I'm just started experimenting with rust for cortex-m and I get a similar problem 
 
    $ rustc -V
    rustc 1.22.0-nightly (417c73891 2017-10-05)

    $ xargo build --example hello
     Compiling r0 v0.2.2
     Compiling cortex-m v0.2.12
     Compiling vcell v0.1.0
     Compiling aligned v0.1.1
     Compiling cortex-m-rt v0.2.5
     Compiling cortex-m-semihosting v0.1.3
     Compiling cc3200 v0.0.1 (file:///home/adona/ZAKKE/rust/cc3200)
     Compiling volatile-register v0.2.0
    error: linking with `arm-none-eabi-ld` failed: exit code: 1

    = note: arm-none-eabi-ld: 
          You must specify the exception handlers.
          Create a non `pub` static variable with type
          `cortex_m::exception::Handlers` and place it in the
          '.rodata.exceptions' section. (cf. #[link_section]). Apply the
          `#[used]` attribute to the variable to make it reach the linker.
          arm-none-eabi-ld: 
          Invalid '.rodata.exceptions' section.
          Make sure to place a static with type `cortex_m::exception::Handlers`
          in that section (cf. #[link_section]) ONLY ONCE.

japaric, 2017-10-07:
This was fixed in cortex-m-rt v0.3.6 so closing. If you still the issue try updating your lockfile.

@attdona That looks like a different issue. The exceptions / interrupts link sections were renamed in cortex-m-rt v0.3.x and you are using v0.2.x. My guess is that you are using a recent svd2rust which only works with cortex-m-rt v0.3.x and cortex-m-quickstart v0.2.x. If you are following one of my first blog post and using cortex-m-quickstart v0.1x then you have to use an older svd2rust releases: v0.7.x is the version used in the blog post (this information is in the blog post). Though I suggest you use cortex-m-quickstart v0.2.x.

closed 2017-10-07 by japaric