rust-embedded/cortex-m-quickstart #35: __RESET_VECTOR has an invalid section specifier

et1975, 2018-06-20:
What am I missing?

```
cargo build --examples
    Updating registry `https://github.com/rust-lang/crates.io-index`
   Compiling cc v1.0.17                                                         
   Compiling vcell v0.1.0
   Compiling aligned v0.2.0
   Compiling bare-metal v0.2.0
   Compiling r0 v0.2.2
   Compiling cortex-m-quickstart v0.3.2 (file:///~/sources/cortex-m-qs)
   Compiling volatile-register v0.2.0
   Compiling cortex-m v0.5.2
   Compiling cortex-m-semihosting v0.3.0
   Compiling cortex-m-rt v0.5.1
LLVM ERROR: Global variable '__RESET_VECTOR' has an invalid section specifier '.vector_table.reset_vector': mach-o section specifier requires a segment whose length is between 1 and 16 characters.
error: Could not compile `cortex-m-rt`.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

japaric, 2018-06-20:
You are (likely) compiling for the host (macOS); you should be cross compiling for one of the THUMB targets e.g. `cargo build --target thumbv7m-none-eabi`, or you can set a default target in .cargo/config.

et1975, 2018-06-20:
Yes, forgot to make sure .cargo/config has a build target. Thank you!

closed 2018-06-20 by et1975