rust-embedded/cortex-m-quickstart #36: No documentation for v0.3.2

gbip, 2018-06-29:
Currently docs.rs [show ](https://docs.rs/cortex-m-quickstart/0.3.2/cortex_m_quickstart/)
> The requested resource does not exist

About `cortex-m-quickstart`.

adamgreig, 2018-08-07:
The problem is [here](https://docs.rs/crate/cortex-m-quickstart/0.3.3/builds/111775) and is an error building `panic-semihosting` on `docs.rs` (which itself doesn't have docs, [same problem](https://docs.rs/crate/panic-semihosting/0.3.0/builds/102433)).

It should clear up once `panic_implementation` becomes stable (rust-lang/rust#44489) (and `panic-semihosting` is updated accordingly), but until then we might be able to hack something to stop docs.rs trying to build it...

korken89, 2019-08-13:
This is almost 1 year old, closing issue, reopen if needed

closed 2019-08-13 by korken89