rust-embedded/cortex-m-quickstart #55: memory.x contains incorrect FLASH : ORIGIN value

wolfgangam, 2018-10-27:
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  /* TODO Adjust these memory regions to match your device memory layout */
  /* These values correspond to the LM3S6965, one of the few devices QEMU can emulate */
  FLASH : ORIGIN = 0x00000000, LENGTH = 256K **<- Value in source code**
  RAM : ORIGIN = 0x20000000, LENGTH = 40K
}

but had to change to 

MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  /* TODO Adjust these memory regions to match your device memory layout */
  /* These values correspond to the LM3S6965, one of the few devices QEMU can emulate */
  FLASH : ORIGIN = **0x08000000**, LENGTH = 256K
  RAM : ORIGIN = 0x20000000, LENGTH = 40K
}

then I was able to load the code onto the microcontroller.

Using Rev C+ of the STM32f3xDiscovery board, but that shouldn't be an issue.

therealprof, 2018-10-27:
@wolfgangam Please note the comment just above the line you've modified. This file is supposed to be adjusted to the MCU you're using and the values are correct for a different MCU.

therealprof, 2018-10-27:
@wolfgangam Actually the README specifically tells you to adjust the file to your needs using the STM32F3DISCOVERY as an example. To be honest I'd have the file contain invalid data so any attempt to compile the file without modification will fail, since this is a step that should never be skipped.

closed 2018-10-27 by wolfgangam