rust-embedded/cortex-m-quickstart #58: cargo build fails: can't find crate for `std`

wucke13, 2018-11-12:
The output of the error is the following:

```
   Compiling rust-toolchain v0.1.0 (/tmp/test)                                                                                                                                       
     Running `rustc --crate-name rust_toolchain src/main.rs --color always --crate-type bin --emit=dep-info,link -C debuginfo=2 -C metadata=f8ada62e29834902 -C extra-filename=-f8ada62e29834902 --out-dir /tmp/test/target/thumbv7em-none-eabihf/debug/deps --target thumbv7em-none-eabihf -C incremental=/tmp/test/target/thumbv7em-none-eabihf/debug/incremental -L dependency=/tmp/test/target/thumbv7em-none-eabihf/debug/deps -L dependency=/tmp/test/target/debug/deps --extern stm32f3=/tmp/test/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-89c96bc170109bc0.rlib -C link-arg=-Tlink.x -L /tmp/test/target/thumbv7em-none-eabihf/debug/build/stm32f3-7c665be50e2d2105/out -L '/home/wucke13/documents/people/Günther Kemnitz/rust-toolchain/target/thumbv7em-none-eabihf/debug/build/cortex-m-e093635255097b24/out' -L /tmp/test/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-28a646978498040b/out`
error[E0463]: can't find crate for `std`                                                                                                                                             
  |                                                                                                                                                                                  
  = note: the `thumbv7em-none-eabihf` target may not be installed                                                                                                                    
                                                                                                                                                                                     
error: aborting due to previous error                                                                                                                                                
                                                                                                                                                                                     
For more information about this error, try `rustc --explain E0463`.                                                                                                                  
error: Could not compile `rust-toolchain`.                                                                                                                                           

Caused by:
  process didn't exit successfully: `rustc --crate-name rust_toolchain src/main.rs --color always --crate-type bin --emit=dep-info,link -C debuginfo=2 -C metadata=f8ada62e29834902 -C extra-filename=-f8ada62e29834902 --out-dir /tmp/test/target/thumbv7em-none-eabihf/debug/deps --target thumbv7em-none-eabihf -C incremental=/tmp/test/target/thumbv7em-none-eabihf/debug/incremental -L dependency=/tmp/test/target/thumbv7em-none-eabihf/debug/deps -L dependency=/tmp/test/target/debug/deps --extern stm32f3=/tmp/test/target/thumbv7em-none-eabihf/debug/deps/libstm32f3-89c96bc170109bc0.rlib -C link-arg=-Tlink.x -L /tmp/test/target/thumbv7em-none-eabihf/debug/build/stm32f3-7c665be50e2d2105/out -L '/home/wucke13/documents/people/Günther Kemnitz/rust-toolchain/target/thumbv7em-none-eabihf/debug/build/cortex-m-e093635255097b24/out' -L /tmp/test/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-28a646978498040b/out` (exit code: 1)
```
`rustup show` shows that the needed target is installed, though:
```
Default host: x86_64-unknown-linux-gnu

installed toolchains
--------------------

stable-x86_64-unknown-linux-gnu
beta-x86_64-unknown-linux-gnu (default)
nightly-x86_64-unknown-linux-gnu

installed targets for active toolchain
--------------------------------------

thumbv6m-none-eabi
thumbv7em-none-eabi
thumbv7em-none-eabihf
thumbv7m-none-eabi
x86_64-unknown-linux-gnu

active toolchain
----------------

beta-x86_64-unknown-linux-gnu (default)
rustc 1.31.0-beta.8 (9142ac95a 2018-11-10)
```
What causes this?

japaric, 2018-12-01:
> error[E0463]: can't find crate for `std`

Likely `src/main.rs` does *not* have the `#![no_std]` attribute in it.

wucke13, 2018-12-24:
Thats it! Thanks!

closed 2018-12-24 by wucke13

H-Y-B, 2019-04-20:
> > error[E0463]: can't find crate for `std`
> 
> Likely `src/main.rs` does _not_ have the `#![no_std]` attribute in it.

But  if i want to print "Hello world" by using std::println! , what should I do?

georgio, 2019-06-07:
@H-Y-B This might answer your question :-)
https://docs.rust-embedded.org/discovery/06-hello-world/index.html

al3x-huang, 2019-10-17:
What if one needs to use other std libraries like std::ptr for example?

adamgreig, 2019-10-18:
> What if one needs to use other std libraries like std::ptr for example?

Everything you _can_ use in a `#[no_std]` context is in `core`, for example: [core::ptr](https://doc.rust-lang.org/core/ptr/index.html). Many `std` modules are re-exported from core.

Niondir, 2020-04-12:
Is there a way to find out where `std` is used? I have the `#![no_std]` in my main.

wucke13, 2020-04-12:
@Niondir Isn't it already stated in the `cargo` output? Other than that, just grep for `std::` in your `src`. Do you have any deps through `Cargo.toml`? Some deps have no_std as a feature, which needs to be invoked explicitly for them to be no_std conform.

Niondir, 2020-04-12:
I actually found another file where the `#![no_std]` was missing.