rust-embedded/cortex-m-quickstart #70: Add a way of unittesting

smilykoch, 2019-06-19:
Would it be possible for someone with experience in such, to add some examples of adding unit testing and possibly integration testing to a project like this?

Maybe even using https://github.com/japaric/utest to allow qemu emulations?

I myself is struggling a lot to work it out, but i don't think it is very simple.. Or at least not very well documented.

korken89, 2019-08-06:
The simple way to do it is to have one "binary" crate and one "firmware" crate. In the firmware crate you can have testing and everything as normal as this crate does not have to be compiled with the thumb targets.

rubberduck203, 2019-11-18:
I actually started working on this at one point.
There's an example of running unit tests on your dev machine in the examples directory.

https://github.com/rust-embedded/cortex-m-quickstart/blob/master/examples/test_on_host.rs

korken89, 2019-11-18:
This is a good way, but if you do want to use the `.cargo/config` file so `cargo build` uses the embedded target directly it can become annoying.
I have a reference here if one wants to compare the 2 approaches and then choose the one that the person likes the best: https://github.com/korken89/rust-embedded-example-project

closed 2024-06-25 by MathiasKoch