rust-embedded/cortex-m-quickstart #73: Uncommenting stm32f3 dependency causes linker error

leshow, 2019-07-24:
setting the runner to:

```
[target.'cfg(all(target_arch = "arm", target_os = "none"))']
runner = "arm-none-eabi-gdb -q -x openocd.gdb"

# and build to
[build]
target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (with FPU)
```
and un-commenting:

```
[dependencies.stm32f3]
features = ["stm32f303", "rt"]
version = "0.7.1"
```

is all that's required to generate a build error (in debug or release):

```
❯ cargo build --release
   Compiling typenum v1.10.0
   Compiling semver-parser v0.7.0
   Compiling proc-macro2 v0.4.30
   Compiling unicode-xid v0.1.0
   Compiling rand_core v0.4.0
   Compiling syn v0.15.41
   Compiling stable_deref_trait v1.1.1
   Compiling cortex-m-rt v0.6.9
   Compiling vcell v0.1.0
   Compiling cortex-m v0.6.0
   Compiling stm32f3 v0.7.1
   Compiling r0 v0.2.2
   Compiling cortex-m-semihosting v0.3.3
   Compiling test v0.1.0 (/home/leshow/dev/embedded/test)
   Compiling panic-halt v0.2.0
   Compiling volatile-register v0.2.0
   Compiling rand_core v0.3.1
   Compiling rand v0.5.6
   Compiling semver v0.9.0
   Compiling generic-array v0.12.3
   Compiling rustc_version v0.2.3
   Compiling as-slice v0.1.0
   Compiling bare-metal v0.2.4
   Compiling aligned v0.3.1
   Compiling quote v0.6.13
   Compiling cortex-m-rt-macros v0.1.5
error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "-L" "/home/leshow/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "/home/leshow/dev/embedded/test/target/thumbv7em-none-eabihf/release/deps/test-1090b9e1da14b251.test.biugehj6-cgu.0.rcgu.o" "-o" "/home/leshow/dev/embedded/test/target/thumbv7em-none-eabihf/release/deps/test-1090b9e1da14b251" "--gc-sections" "-L" "/home/leshow/dev/embedded/test/target/thumbv7em-none-eabihf/release/deps" "-L" "/home/leshow/dev/embedded/test/target/release/deps" "-L" "/home/leshow/dev/embedded/test/target/thumbv7em-none-eabihf/release/build/test-bd5b373cb2319a3a/out" "-L" "/home/leshow/dev/embedded/test/target/thumbv7em-none-eabihf/release/build/cortex-m-5aa8e37429c9abc2/out" "-L" "/home/leshow/dev/embedded/test/target/thumbv7em-none-eabihf/release/build/cortex-m-rt-3a1a97e3c2d417dd/out" "-L" "/home/leshow/dev/embedded/test/target/thumbv7em-none-eabihf/release/build/cortex-m-semihosting-35622e2bd8b57563/out" "-L" "/home/leshow/dev/embedded/test/target/thumbv7em-none-eabihf/release/build/stm32f3-fd1a3010711a7fc0/out" "-L" "/home/leshow/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/tmp/rustcs88vjN/libcortex_m_rt-880bf24c42db6019.rlib" "/tmp/rustcs88vjN/libcortex_m-b41d19b44405705d.rlib" "--start-group" "--end-group" "/home/leshow/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-d51751fd2e6fcbd1.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
```

If I follow the advice in `config` and also uncomment:

` "-C", "linker=arm-none-eabi-ld"`

I get an error with that linker:

```
= note: arm-none-eabi-ld: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
```

adamgreig, 2019-07-24:
Hi, thanks for reporting this bug!

The specific problem is that while the stm32f3 crate does provide the required interrupt vectors,  `cargo build` just builds `src/main.rs`, which does not include stm32f3 (no `use` or `extern crate` line), and as such the stm32f3 crate isn't linked in, so the interrupt vectors don't make it either.

The documentation suggests actually building the `device` example with this feature, which should work, but hasn't actually been updated when the `.cargo/config` was, so tries to pull in a different device crate.

If you modify `examples/device.rs` as follows:

```dif
-use stm32f30x::{interrupt, Interrupt, NVIC};
+use stm32f3::stm32f303::{interrupt, Interrupt, NVIC};
```

then you should be able to run `cargo build --example device` without errors.

Obviously this is still a bug, so I'll get it fixed ASAP, thanks again for reporting!

leshow, 2019-07-24:
I can submit a PR with that change if you like

adamgreig, 2019-07-24:
Thanks for offering! I have just opened #74 to fix it though.

leshow, 2019-07-24:
I figure I'll ask while you're here, but trying to build anything else except device while the dependency is there will fail in a similar linker error. Is there any way to fix that? Or is the answer to use something like the above import in that situation also.

adamgreig, 2019-07-24:
You might want to check out https://github.com/adamgreig/stm32f4-demo which is my barebones example of an embedded stm32 application.

commit d018974c (2019-08-06): Update instructions for device example, closes #73
commit f5a87210 (2019-08-06): Merge pull request #74 from rust-embedded/fix-73

Update instructions for device example, closes #73
closed 2019-08-06 by korken89