rust-embedded/cortex-m-quickstart #76: Update panic-semihosting version in generated Cargo.toml

bobgates, 2019-08-27:
At the moment, cortex-m-quickstart generates a Cargo.toml that includes a dependency on panic-semihosting version 0.3.0

This leads to an error when building:

```   Compiling panic-semihosting v0.3.0
error[E0557]: feature has been removed
  --> /Users/declan/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-semihosting-0.3.0/src/lib.rs:59:12
   |
59 | #![feature(panic_implementation)]
   |            ^^^^^^^^^^^^^^^^^^^^
   |
note: subsumed by `#[panic_handler]` ```

If the version of panic-semihosting is changed to 0.5.2, then the generated program builds as per the instructions in the documentation.

RamArjunVarma, 2020-09-22:
hi sir How to change the version can you explain please

RamArjunVarma, 2020-09-26:
ok sir will try that thanks for the reply