rust-embedded/cortex-m-quickstart #85: VSCode GDB Launch Configuration Not Taken From `openocd.gdb`

david-boles, 2020-03-06:
Would it be possible to configure the VSCode `cortex-debug` extension to pull its gdb commands from `openocd.gdb`? I ran into this when trying to use semihosting, `openocd.gdb` enables it but the `cortex-debug` launch configuration doesn't. If not (I couldn't see how), should I make a pull request to add `"preLaunchCommands": ["monitor arm semihosting enable"]` to the launch configurations?

rubberduck203, 2020-03-10:
AFAIK it’s not currently possible to load a gdb script through the cortex-debug extension. I would encourage you to [open an issue there](https://github.com/Marus/cortex-debug/issues) as I couldn’t find a related one. 

I think it’s reasonable to open a PR that adds semihosting to the QEMU launch config, since the stm32f303 config is set up to use ITM. That way we have an example for both communication mechanisms.

andresv, 2020-05-07:
You can add this to your `launch.json`. Worked nicely with stm32g0.
```
            "postLaunchCommands": [
                "monitor arm semihosting enable",
            ],
```

Also `cortex-debug` supports a lot of parameters that can be tweaked: https://github.com/Marus/cortex-debug/blob/master/package.json#L195

andresv, 2020-05-07:
Damn, I should have read OP more carefully. It was all there.

rubberduck203, 2021-06-17:
Can this issue be closed?

reitermarkus, 2022-04-01:
Ah, nevermind, `hprintln` output is in the `gdb-server` terminal.