rust-embedded/cortex-m-quickstart #88: Blue Pill Quickstart

reneherrero, 2020-06-07:
Just wanted to say thanks for putting this out there, was a tremendous help.

I derived a Blue Pill Quickstart from it with VS Code debugging working great: https://github.com/reneherrero/blue-pill-quickstart

I though about doing a PR to give something back, but I don't see how to do so without creating clutter... lots of possible permutations in this area.

Anyhow, let me know if you're interested and I'll try to cook something up.

Again, thanks!

rubberduck203, 2020-06-07:
Nice!

I’m not part of the team, but I’m excited to see this and believe there should be many templates out there to make it easy for folks to get started on a particular target. Your repo is a nice addition to the ecosystem. 

You should consider adding an announcement to the Rust embedded blog. 
https://github.com/rust-embedded/blog

reneherrero, 2020-06-07:
Agree it would be helpful to have more templates out there, getting started in the embedded space is most definitely several orders of magnitude more complicated than when coding on top of an OS.  I'm surprised it works at all considering how many moving parts are involved!

The only little annoying thing with what I put together is the VS Code Rust add-in thinks there's an error with the toml when in fact there isn't.  From the looks of it, it's doing it's own build somewhere and that is failing.  But it's not a show stopper.  I mentioned it [here](https://github.com/rust-lang/rls-vscode/issues/730).

This project looks intended to work with `cargo generate`, which I'm not very familiar with, but it does seem to impose pretty strict limitations.  Doesn't seem ideal for creating multiple "flavors" from a single repo.

rubberduck203, 2020-06-07:
Yeah. There’s an open issue to allow more flexibility in that area. https://github.com/ashleygwilliams/cargo-generate/issues/17

therealprof, 2020-06-07:
We should probably add them all to the awesome list... https://github.com/rust-embedded/awesome-embedded-rust

Maybe even add a separate section to collect all the quickstart templates.

reneherrero, 2020-06-07:
The way I see it, having a working setup is two fold:
1. You need to create a project -> A `template.rs` in `cargo generate` would be ideal as it would be way easier to keep it up to date than having multiple quickstart projects scattered around with multiple owners. Would probably go stale fast.
2. You need to setup your hardware (ex: properly connecting ST-Link v2 to the board) and dependencies (ex: openocd, gdb-multi-arch, udev rules, etc.) -> Tough nut to crack as there's lots permutations out there there.

Having it all here would be ideal, but that would require a big effort.  In the meantime, @therealprof's suggestion is probably the best option.