rust-embedded/cortex-m-quickstart #93: Update for stm32l4 dependency causes linker error

falloutkid, 2020-08-09:
I update follwing files for using STM32L476

- Cargo.toml

```rust
[dependencies.stm32l4]
features = ["stm32l4x6", "rt"]
version = "0.11.0"
```

- config

```
[build]
# Pick ONE of these compilation targets
# target = "thumbv6m-none-eabi"    # Cortex-M0 and Cortex-M0+
# target = "thumbv7m-none-eabi"    # Cortex-M3
# target = "thumbv7em-none-eabi"     # Cortex-M4 and Cortex-M7 (no FPU)
target = "thumbv7em-none-eabihf" # Cortex-M4F and Cortex-M7F (with FPU)
```

- memory.x

```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  /* TODO Adjust these memory regions to match your device memory layout */
  /* These values correspond to the LM3S6965, one of the few devices QEMU can emulate */
  FLASH : ORIGIN = 0x08000000, LENGTH = 1024K
  RAM : ORIGIN = 0x20000000, LENGTH = 96K
}
```

- examples/device.rs

```rust
use stm32l4::stm32l4x6::{interrupt, Interrupt, NVIC};
```

Then, I compline device `cargo build --example device` is finish.  After that, I complie hello 'cargo build --example hello'. But it happen a build error.

```
/mnt/rust/src/app# cargo build --example hello
   Compiling awesome-app v0.1.0 (/mnt/rust/src/app)
warning: unused import: `debug`
 --> examples/hello.rs:9:28
  |
9 | use cortex_m_semihosting::{debug, hprintln};
  |                            ^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

error: linking with `rust-lld` failed: exit code: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "-L" "/usr/local/rustup/toolchains/1.45.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/examples/hello-8f9cd6364696548c.2qm8sestblbxkequ.rcgu.o" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/examples/hello-8f9cd6364696548c.2x4nait49zpt1oyp.rcgu.o" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/examples/hello-8f9cd6364696548c.3fqpxmhbeby0vnr9.rcgu.o" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/examples/hello-8f9cd6364696548c.5cvtkejs45hfcola.rcgu.o" "-o" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/examples/hello-8f9cd6364696548c" "--gc-sections" "-L" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps" "-L" "/mnt/rust/src/app/target/debug/deps" "-L" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/build/awesome-app-b6d72e815af4ef45/out" "-L" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/build/cortex-m-25ad3822f8a7a2a8/out" "-L" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-dc75f42577b5d9d9/out" "-L" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/build/cortex-m-semihosting-6a1c39732392316c/out" "-L" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/build/stm32l4-6cc22a755524db5d/out" "-L" "/usr/local/rustup/toolchains/1.45.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib" "-Bstatic" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_semihosting-41ed8b6732571890.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-4a546d3f8fe96189.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-931097aba76b036e.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libvcell-5ad610f7cd60d5ba.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-1389f903efc94440.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libaligned-9a89ddbd462d1e07.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libas_slice-377848fecaa59fea.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libstable_deref_trait-a11f983656b18d1d.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-b93168bf128964f5.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libgeneric_array-b284d275589af3fb.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libtypenum-fc489d0bc1f0e902.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libcortex_m_rt-3e88d67a44ef2c5c.rlib" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libr0-2ddfb6c5c3292139.rlib" "--start-group" "/mnt/rust/src/app/target/thumbv7em-none-eabihf/debug/deps/libpanic_halt-d35b0c7ceab6b642.rlib" "/usr/local/rustup/toolchains/1.45.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/librustc_std_workspace_core-3370a0a65a1e95ca.rlib" "/usr/local/rustup/toolchains/1.45.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-688cc81889ae99dc.rlib" "--end-group" "/usr/local/rustup/toolchains/1.45.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-7b0ac0a4cad7c4db.rlib" "-Tlink.x" "-Bdynamic"
  = note: rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
          
          rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
          
          rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
          

error: aborting due to previous error; 1 warning emitted
```

Is this sorce code support STM32L476?

thalesfragoso, 2020-08-09:
The problem is that the PAC itself isn't really used in the example and the PAC is the responsible for providing the interrupt vectors to `cortex-m-rt`, the simplest solution is to have something like this:
```rust
use stm32l4 as _;
```

But yeah, the example is wrong and should be fixed.

therealprof, 2020-08-09:
Hm, usually that should work automatically via the prelude. If it doesn't that's probably a problem with the HAL impl.

thalesfragoso, 2020-08-09:
> Hm, usually that should work automatically via the prelude. If it doesn't that's probably a problem with the HAL impl.

The problem is that the example in question doesn't use the prelude, or any use statements for the PAC.