rust-embedded/cortex-m-quickstart #97: Documentation or example for the allocator not working

huntc, 2020-10-08:
The documentation appears to be out of date. If I follow the steps then, to build the allocator example, I get the following error:

```
% cargo build --example allocator         
   Compiling typenum v1.12.0
   Compiling version_check v0.9.2
   Compiling semver-parser v0.7.0
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling cortex-m v0.6.3
   Compiling syn v1.0.42
   Compiling stable_deref_trait v1.2.0
   Compiling vcell v0.1.2
   Compiling cortex-m-semihosting v0.3.5
   Compiling volatile-register v0.1.2
   Compiling bitfield v0.13.2
   Compiling cortex-m-rt v0.6.13
   Compiling test v0.1.0 (/Users/huntc/Desktop/test/test)
   Compiling r0 v0.2.2
   Compiling linked_list_allocator v0.6.6
error[E0554]: `#![feature]` may not be used on the stable release channel
 --> /Users/huntc/.cargo/registry/src/github.com-1ecc6299db9ec823/linked_list_allocator-0.6.6/src/lib.rs:1:1
  |
1 | #![feature(const_fn)]
  | ^^^^^^^^^^^^^^^^^^^^^
...
```

If I try with the nightly:

```
huntc@Christophers-MacBook-Pro test % cargo +nightly build --example allocator
   Compiling typenum v1.12.0
   Compiling semver-parser v0.7.0
   Compiling version_check v0.9.2
   Compiling proc-macro2 v1.0.24
   Compiling unicode-xid v0.2.1
   Compiling cortex-m v0.6.3
   Compiling syn v1.0.42
   Compiling stable_deref_trait v1.2.0
   Compiling vcell v0.1.2
   Compiling cortex-m-semihosting v0.3.5
   Compiling volatile-register v0.1.2
   Compiling cortex-m-rt v0.6.13
   Compiling bitfield v0.13.2
   Compiling linked_list_allocator v0.6.6
   Compiling test v0.1.0 (/Users/huntc/Desktop/test/test)
   Compiling r0 v0.2.2
error[E0432]: unresolved import `alloc::alloc::AllocErr`
  --> /Users/huntc/.cargo/registry/src/github.com-1ecc6299db9ec823/linked_list_allocator-0.6.6/src/lib.rs:14:30
   |
14 | use alloc::alloc::{AllocRef, AllocErr, Layout};
   |                              ^^^^^^^^
   |                              |
   |                              no `AllocErr` in `alloc`
   |                              help: a similar name exists in the module: `AllocError`
...
```

Any pointers no how to see that the allocator example is able to be built? Is the documentation/example out of date?

thalesfragoso, 2020-10-11:
Hmm, could you try updating this line to `0.4.0`:

https://github.com/rust-embedded/cortex-m-quickstart/blob/be44af69cc91e2ff5b75f400e09488cba44e9f59/Cargo.toml#L18

huntc, 2020-10-11:
Thanks @thalesfragoso . I did that, but then receive:

```
% cargo build --example allocator
   Compiling linked_list_allocator v0.8.6
   Compiling typenum v1.12.0
   Compiling semver v0.9.0
   Compiling generic-array v0.14.4
   Compiling proc-macro2 v1.0.24
   Compiling cortex-m v0.6.3
   Compiling syn v1.0.42
   Compiling cortex-m-rt v0.6.13
error[E0554]: `#![feature]` may not be used on the stable release channel
 --> /Users/huntc/.cargo/registry/src/github.com-1ecc6299db9ec823/linked_list_allocator-0.8.6/src/lib.rs:1:1
  |
1 | #![feature(const_mut_refs)]
  | ^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0554`.
error: could not compile `linked_list_allocator`.

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

PR #98 (2020-10-13): Fix alloc example
Fixes #97
closed 2020-10-13 by therealprof