rust-embedded/cortex-m-quickstart #99: Feature request: Confusing task names

bradleyharden, 2020-11-05:
Hi,

I didn't have much experience with VSCode before using it with Rust. I was working with Cortex-Debug and the `preLaunchTask` field. The current Rust task names look like shell commands. Because of that, I assumed I could change the `preLaunchTask` string to a different shell command and it would work. I only realized later that the `preLaunchTask` string is actually a task name, and that the task names just happen to look like shell commands.

I think this is misleading and probably a source of confusion for others who are not familiar with VSCode tasks. Can I suggest changing the names to something that doesn't look like a shell command?

rubberduck203, 2020-11-05:
I’m open to suggestions if you have better names in mind. 

I assume you’re talking about these labels?

https://github.com/rust-embedded/cortex-m-quickstart/blob/3cbcb9956acd638c0fb864937770f51d663fba6f/.vscode/tasks.json#L12

bradleyharden, 2020-11-07:
@rubberduck203, yes that's what I'm talking about. I didn't have anything specific in mind. I just want something that looks like a label rather than a shell command. Maybe `Cargo - Build` and `Cargo - Build (release)`, or something like that? It's also not that big of a deal. I figured it out. It was just a bit confusing.

rubberduck203, 2020-11-07:
How about the following?

- Cargo Build (debug)
- Cargo Build (release)
- Cargo Build Examples (debug)
- Cargo Build Examples (release)
- Cargo Clean

bradleyharden, 2020-11-07:
Those look good to me.

There's no rush. I'll also try to get around to it soon, but you might beat me. Thanks for considering the change.

rubberduck203, 2020-11-09:
@bradleyharden PR is open. Just need to wait for a maintainer now.
Thanks for bringing it to my attention.

closed 2020-11-09 by adamgreig

bradleyharden, 2020-11-09:
Thanks @rubberduck203!

commit 18bb6807 (2020-11-09): Merge pull request #100 from rubberduck203/master

Fixes #99: Disambiguates VSCode Tasks
commit eb68e8bb (2020-11-09): Fixes #99: Disambiguates VSCode Tasks

Using the equivalent command line as the task name caused some confusion.
Users were under the impression that they could use any valid cargo command as a preLaunchTask in the launch configurations.