rust-embedded/cortex-m-quickstart #107: Using probe-rs to initialise memory.x

romixlab, 2021-06-30:
Wouldn't it be great to automatically create a proper linker configuration based on the selected chip? It seems that this information can be easily pulled from yaml files in `probe-rs/targets/`. Probably someone is already thought about it, in that case just close this right away.

rubberduck203, 2021-06-30:
I’m fairly certain there would need to be changes to cargo generate to make something like this work. 

https://github.com/cargo-generate/cargo-generate/issues/17

romixlab, 2021-06-30:
Some way of getting values out of yaml files will also be required, maybe it's easier to create a cargo tool tailored specifically for that task...

rubberduck203, 2021-06-30:
Oh! Hey! Some of the required functionality landed in cargo-generate v0.6.0!

https://github.com/cargo-generate/cargo-generate#template-defined-placeholders

romixlab, 2021-07-02:
Great! That can be used for improvements.

As for the memory size part, probably some scripting functionality from cargo-generate will be required. I wonder what would be better, add scripting to cargo-generate or create a special tool specifically for creating embedded projects.

rubberduck203, 2021-07-02:
I don’t know much about probe-rs. 
Couldn’t we use the same user provided value for memory.x and the probe-rs config?

Of course, that assumes adding probe-rs support to the quickstart.