rust-embedded/cortex-m-quickstart #110: Make this repository a template

nbstrong, 2021-12-05:
Github has template support now. Please make this repository one.

adamgreig, 2021-12-05:
Currently this repository is designed to be used with `cargo generate`, which means there are a bunch of placeholder values which are not valid if you just copy the repository (mainly in `Cargo.toml`).

Making it a GitHub template is a nice idea, but it might be annoying that it wouldn't work out the box without replacing all the things `cargo generate` replaces, and if someone uses `cargo generate`, they probably don't need it to be a GitHub template, since `cargo generate` will copy the files for them anyway.

I've not used `cargo generate` myself though, maybe it would be more convenient to just be a GitHub template?

nbstrong, 2021-12-05:
It really would and imo I'd prefer to have one less tool installed on my system I have to figure out. Furthermore, you'd gain having a reference back to the upstream you can pull updates from.

The only issue is pointing the project variables to the app name.

The less ideal solution would be to run a script that changes them. Id prefer it automatically infer it from the base git repo name which you set when forking from the template repository.

nbstrong, 2021-12-05:
These appear to be all the placeholders
[image]

nbstrong, 2021-12-05:
This post seemed to discuss a solution by combining Github Actions:
https://stefanbuck.com/blog/repository-templates-meets-github-actions

I am working on a prototype.