rust-embedded/cortex-m-quickstart #112: "message": "failed to parse manifest at `/Users/tz/Downloads/cortex-m-quickstart-master/Cargo.toml`",

taozhiwei21, 2022-03-11:
tz@taodeMacBook-Pro cortex-m-quickstart-master % cargo build --target thumbv7m-none-eabi
error: failed to parse manifest at `/Users/tz/Downloads/cortex-m-quickstart-master/Cargo.toml`

Caused by:
  invalid character `{` in package name: `{{project-name}}`, the first character must be a Unicode XID start character (most letters or `_`)
tz@taodeMacBook-Pro cortex-m-quickstart-master % 

"/Users/tz/Downloads/cortex-m-quickstart-master/Cargo.toml", 	"owner": "rust-file:///Users/tz/Downloads/cortex-m-quickstart-master", 	"severity": 8, 	"message": "failed to parse manifest at `/Users/tz/Downloads/cortex-m-quickstart-master/Cargo.toml`", 	"startLineNumber": 1, 	"startColumn": 1, 	"endLineNumber": 10000, 	"endColumn": 1 }]

newAM, 2022-03-11:
It looks like that repository was a direct clone, that field usually gets filled by cargo generate.

Try this:

`cargo generate --git https://github.com/rust-embedded/cortex-m-quickstart`

Install instructions for cargo generate are in the README 👍

newAM, 2022-03-12:
Did that work for you @taozhiwei21 ?  Happy to help if you run into more problems.