rust-embedded/cortex-m-quickstart #116: add a runner to flash program directly

Narukara, 2023-01-14:
Can we add a runner to flash program directly through openocd? It can be annoying to start openocd and close gdb manually every time you flash.

In fact, I did this in my own project:

1. Add `runner = "./flash.sh"` to `config.toml`
2. Create a `flash.sh` in the project root directory, and its contents are:

```shell
openocd -f openocd.cfg -c init -c halt -c "flash write_image erase $1" -c reset -c shutdown
```

Then I can flash the program directly with `cargo run`. This may not be a good way to implement it, but it's really convenient.

crespum, 2023-04-02:
Are you familiar with [cargo-flash](https://probe.rs/docs/tools/cargo-flash/)? I was doing something similar to what you propose but now I simply use something like `cargo flash --release --chip NRF52820_XXAA`. It works like a charm!

Narukara, 2023-04-03:
> Are you familiar with [cargo-flash](https://probe.rs/docs/tools/cargo-flash/)? I was doing something similar to what you propose but now I simply use something like `cargo flash --release --chip NRF52820_XXAA`. It works like a charm!

I tried cargo-flash as you said and it works perfectly. Thanks for your advice!

Now that there is a tool available, I think this issue can be closed.

closed 2023-04-03 by Narukara