rust-embedded/cortex-m-quickstart #121: Error while following the Embedded Book

dioguerra, 2023-05-07:
Qemu section `Using cargo-generate`

While creating project from template I get an error:

```
$ cargo install cargo-generate --git https://github.com/rust-embedded/cortex-m-quickstart
    Updating git repository `https://github.com/rust-embedded/cortex-m-quickstart`
error: failed to parse manifest at `/home/dtomasgu/.cargo/git/checkouts/cortex-m-quickstart-e5c41c227d09013e/6b2fd0e/Cargo.toml`

Caused by:
  invalid character `{` in package name: `{{project-name}}`, the first character must be a Unicode XID start character (most letters or `_`)

$ cargo --version
cargo 1.68.1 (115f34552 2023-02-26)
```

thalesfragoso, 2023-05-09:
I think you're misunderstanding how `cargo-generate` works. First you install it with `cargo install`, then you use it with `cargo generate`. The book does these two steps, btw.

dioguerra, 2023-05-09:
I tried it again and it worked. Not sure what was wrong. Eitherway, the first time I git cloned and replaced directly

closed 2023-05-09 by dioguerra