rust-embedded/cortex-m-quickstart #126: Instructions appear to be wrong

inferiorhumanorgans, 2024-05-31:
https://docs.rust-embedded.org/cortex-m-quickstart/cortex_m_quickstart/

While following step 2:

(**334** build dependencies to avoid running `git clone` later… *yikes*)

```
$ cargo clone cortex-m-quickstart --vers 0.3.4
error: unexpected argument '--vers' found

  tip: a similar argument exists: '--version'
  tip: to pass '--vers' as a value, use '-- --vers'

Usage: cargo clone --version <--color <COLORING>|-v|-q|--registry <REGISTRY>|--index <URL>|--local-registry <PATH>|--git|CRATE|DIRECTORY>

For more information, try '--help'.
```

newAM, 2024-06-01:
How did you find that page?

You are right, it is out of date.  The README appears to be accurate though.  I didn't know we had a docs.rust-embedded.org.  Should update it or remove it, but I'm curious how you got to that page.

inferiorhumanorgans, 2024-06-05:
IIRC search engine either DDG or Google.